jcchurch13/Mechaduino-Firmware #4: Debugging units: No movement in closed loop

SloMusti, 2016-09-21:
Self-built Mechaduino while waiting for Kickstarter units to arrive. Performed calibration routine and programmed it. Step mode with "s" command works. After sending "y" and "x" and "r 1080" no rotation occurs. Stepper axis can be easily moved my hand, there is some minor resistance, but no closed-loop control action occurs.

What parameters in the firmware to observe or print to evaluate what is going on.

jcchurch13, 2016-09-21:
Sounds like this may be an encoder issue.  Did the calibration routine yield a lookup table with numbers ascending by about 0.02?

SloMusti, 2016-09-22:
Please see the calibration values below, seems quite ok in the way the values are increasing. I have tried with different magnets and motors, but the problem persisted. I doubt the problem is in the H bridge or so as it works fine in the open loop step mode. But as soon as it gets into the closed loop mode, there is almost no torque on the axis.

`60.01 , 60.05 , 60.10 , 60.14 , 60.19 , 60.23 , 60.27 , 60.32 , 60.36 , 60.40 , 60.45 , 60.49 , 60.54 , 60.58 , 60.62 , 60.67 , 60.71 , 60.76 , 60.80 , 60.84 , 60.89 , 60.93 , 60.98 , 61.02 , 61.06 , 61.11 , 61.15 , 61.19 , 61.24 , 61.28 , 61.33 , 61.37 , 61.41 , 61.46 , 61.50 , 61.55 , 61.59 , 61.63 , 61.68 , 61.72 , 61.77 , 61.78 , 61.83 , 61.87 , 61.92 , 61.96 , 62.01 , 62.05 , 62.10 , 62.14 , 62.19 , 62.23 , 62.28 , 62.32 , 62.37 , 62.41 , 62.46 , 62.50 , 62.55 , 62.59 , 62.64 , 62.68 , 62.73 , 62.77 , 62.82 , 62.86 , 62.91 , 62.95 , 63.00 , 63.04 , 63.09 , 63.13 , 63.18 , 63.22 , 63.27 , 63.31 , 63.36 , 63.40 , 63.45 , 63.49 , 63.54 , 63.61 , 63.65 , 63.70 , 63.74 , 63.79 , 63.83 , 63.87 , 63.92 , 63.96 , 64.00 , 64.05 , 64.09 , 64.14 , 64.18 , 64.22 , 64.27 , 64.31 , 64.36 , 64.40 , 64.44 , 64.49 , 64.53 , 64.58 , 64.62 , 64.66 , 64.71 , 64.75 , 64.79 , 64.84 , 64.88 , 64.93 , 64.97 , 65.01 , 65.06 , 65.10 , 65.15 , 65.19 , 65.23 , 65.28 , 65.32 , 65.37 , 65.51 , 65.56 , 65.60 , 65.64 , 65.68 , 65.72 , 65.76 , 65.79 , 65.83 , 65.87 , 65.92 , 65.96 , 66.00 , 66.04 , 66.08 , 66.11 , 66.15 , 66.19 , 66.24 , 66.28 , 66.32 , 66.36 , 66.40 , 66.43 , 66.47 , 66.51 , 66.56 , 66.60 , 66.64 , 66.68 , 66.72 , 66.76 , 66.79 , 66.83 , 66.87 , 66.92 , 66.96 , 67.00 , 67.04 , 67.08 , 67.11 , 67.15 , 67.19 , 67.24 , 67.28 , 67.26 , 67.31 , 67.35 , 67.39 , 67.43 , 67.47 , 67.52 , 67.56 , 67.60 , 67.64 , 67.68 , 67.73 , 67.77 , 67.81 , 67.85 , 67.89 , 67.93 , 67.98 , 68.02 , 68.06 , 68.10 , 68.14 , 68.19 , 68.23 , 68.27 , 68.31 , 68.35 , 68.40 , 68.44 , 68.48 , 68.52 , 68.56 , 68.60 , 68.65 , 68.69 , 68.73 , 68.77 , 68.81 , 68.86 , 68.90 , 68.94 , 68.98 , 69.02 , 68.95 , 69.00 , 69.04 , 69.09 , 69.13 , 69.18 , 69.23 , 69.27 , 69.32 , 69.36 , 69.41 , 69.46 , 69.50 , 69.55 , 69.60 , 69.64 , 69.69 , 69.73 , 69.78 , 69.83 , 69.87 , 69.92 , 69.96 , 70.01 , 70.06 , 70.10 , 70.15 , 70.19 , 70.24 , 70.29 , 70.33 , 70.38 , 70.43 , 70.47 , 70.52 , 70.56 , 70.61 , 70.66 , 70.70 , 70.81 , 70.85 , 70.90 , 70.94 , 70.99 , 71.03 , 71.07 , 71.12 , 71.16 , 71.20 , 71.25 , 71.29 , 71.34 , 71.38 , 71.42 , 71.47 , 71.51 , 71.56 , 71.60 , 71.64 , 71.69 , 71.73 , 71.78 , 71.82 , 71.86 , 71.91 , 71.95 , 72.00 , 72.04 , 72.08 , 72.13 , 72.17 , 72.21 , 72.26 , 72.30 , 72.35 , 72.39 , 72.43 , 72.48 , 72.52 , 72.57 , 72.71 , 72.75 , 72.79 , 72.83 , 72.87 , 72.92 , 72.96 , 72.99 , 73.03 , 73.07 , 73.11 , 73.15 , 73.19 , 73.24 , 73.28 , 73.31 , 73.35 , 73.39 , 73.43 , 73.47 , 73.51 , 73.56 , 73.60 , 73.63 , 73.67 , 73.71 , 73.75 , 73.79 , 73.83 , 73.87 , 73.92 , 73.96 , 73.99 , 74.03 , 74.07 , 74.11 , 74.15 , 74.19 , 74.24 , 74.28 , 74.31 , 74.35 , 74.39 , 74.43 , 74.47 , 74.49 , 74.53 , 74.57 , 74.61 , 74.65 , 74.69 , 74.74 , 74.78 , 74.82 , 74.86 , 74.90 , 74.94 , 74.98 , 75.02 , 75.06 , 75.10 , 75.15 , 75.19 , 75.23 , 75.27 , 75.31 , 75.35 , 75.39 , 75.43 , 75.47 , 75.51 , 75.55 , 75.60 , 75.64 , 75.68 , 75.72 , 75.76 , 75.80 , 75.84 , 75.88 , 75.92 , 75.96 , 76.00 , 76.04 , 76.09 , 76.13 , 76.17 , 76.21 , 76.25 , 76.18 , 76.23 , 76.27 , 76.32 , 76.36 , 76.41 , 76.45 , 76.50 , 76.54 , 76.59 , 76.63 , 76.68 , 76.72 , 76.77 , 76.81 , 76.86 , 76.90 , 76.95 , 76.99 , 77.04 , 77.08 , 77.13 , 77.17 , 77.22 , 77.26 , 77.31 , 77.35 , 77.40 , 77.44 , 77.49 , 77.53 , 77.58 , 77.62 , 77.67 , 77.71 , 77.76 , 77.80 , 77.85 , 77.89 , 77.94 , 78.04 , 78.08 , 78.12 , 78.17 , 78.21 , 78.25 , 78.29 , 78.34 , 78.38 , 78.42 , 78.47 , 78.51 , 78.55 , 78.60 , 78.64 , 78.68 , 78.72 , 78.77 , 78.81 , 78.85 , 78.89 , 78.94 , 78.98 , 79.02 , 79.07 , 79.11 , 79.15 , 79.19 , 79.24 , 79.28 , 79.32 , 79.37 , 79.41 , 79.45 , 79.50 , 79.54 , 79.58 , 79.62 , 79.67 , 79.71 , 79.75 , 79.79 , 79.89 , 79.93 , 79.97 , 80.01 , 80.05 , 80.10 , 80.14 , 80.18 , 80.22 , 80.26 , 80.30 , 80.34 , 80.38 , 80.42 , 80.46 , 80.50 , 80.55 , 80.59 , 80.63 , 80.67 , 80.71 , 80.75 , 80.79 , 80.83 , 80.87 , 80.91 , 80.95 , 81.00 , 81.04 , 81.08 , 81.12 , 81.16 , 81.20 , 81.24 , 81.28 , 81.32 , 81.36 , 81.40 , 81.44 , 81.49 , 81.53 , 81.57 , 81.61 , 81.65 , 81.69 , 81.73 , 81.77 , 81.81 , 81.85 , 81.89 , 81.94 , 81.98 , 82.02 , 82.06 , 82.10 , 82.14 , 82.18 , 82.22 , 82.26 , 82.30 , 82.35 , 82.39 , 82.43 , 82.47 , 82.51 , 82.55 , 82.59 , 82.63 , 82.67 , 82.71 , 82.75 , 82.80 , 82.84 , 82.88 , 82.92 , 82.96 , 83.00 , 83.04 , 83.08 , 83.12 , 83.16 , 83.20 , 83.24 , 83.29 , 83.33 , 83.37 , 83.41 , 83.45 , 83.44 , 83.48 , 83.52 , 83.57 , 83.61 , 83.65 , 83.70 , 83.74 , 83.78 , 83.82 , 83.87 , 83.91 , 83.95 , 84.00 , 84.04 , 84.08 , 84.12 , 84.17 , 84.21 , 84.25 , 84.30 , 84.34 , 84.38 , 84.42 , 84.47 , 84.51 , 84.55 , 84.60 , 84.64 , 84.68 , 84.72 , 84.77 , 84.81 , 84.85 , 84.90 , 84.94 , 84.98 , 85.02 , 85.07 , 85.11 , 85.15 , 85.20 , 85.24 , 85.28 , 85.32 , 85.37 , 85.41 , 85.45 , 85.49 , 85.54 , 85.58 , 85.62 , 85.67 , 85.71 , 85.75 , 85.80 , 85.84 , 85.88 , 85.92 , 85.97 , 86.01 , 86.05 , 86.09 , 86.14 , 86.18 , 86.22 , 86.27 , 86.31 , 86.35 , 86.39 , 86.44 , 86.48 , 86.52 , 86.57 , 86.61 , 86.65 , 86.70 , 86.74 , 86.78 , 86.82 , 86.87 , 86.91 , 86.95 , 86.99 , 87.06 , 87.11 , 87.15 , 87.19 , 87.23 , 87.27 , 87.32 , 87.36 , 87.40 , 87.44 , 87.48  [... cut ...]  57.51 , 57.55 , 57.59 , 57.64 , 57.68 , 57.73 , 57.77 , 57.81 , 57.86 , 57.90 , 57.95 , 57.99 , 58.03 , 58.08 , 58.12 , 58.17 , 58.34 , 58.38 , 58.42 , 58.46 , 58.50 , 58.53 , 58.57 , 58.61 , 58.65 , 58.69 , 58.73 , 58.77 , 58.81 , 58.85 , 58.89 , 58.93 , 58.96 , 59.00 , 59.04 , 59.08 , 59.12 , 59.16 , 59.20 , 59.24 , 59.28 , 59.32 , 59.36 , 59.40 , 59.43 , 59.47 , 59.51 , 59.55 , 59.59 , 59.63 , 59.67 , 59.71 , 59.75 , 59.79 , 59.83 , 59.86 , 59.90 , 59.94 , 59.98 , 60.02 , 60.06 , 60.10 ,  
`

jcchurch13, 2016-09-22:
Looks to me like you may be using a motor with a step angle other than 1.8 degrees (200 steps per rev).  If so, you will need to change the value of "spr" (steps per rev) in parameters.cpp.  

Your cal table should increment by roughly 0.02 and wrap to 0 through after 360.  Your appears to increment by roughly 0.04 and it stalls around 360.

SloMusti, 2016-09-22:
I have checked that, but the motor actually makes a 360deg turn with 200 steps. I have tried the 400 step motor as well and changed the firmware accordingly, but the same problem was present.

Testing as on NEMA17 200step/rotation and on NEMA17 400step/rotation. Different magnet setup, both having sufficient field.

Could be some odd hardware fault.

jcchurch13, 2016-09-23:
Hmm... if you post the raw encoder readings that are printed during the cal routine, that might give me a better idea of whats going wrong.

SloMusti, 2016-09-24:
Will do that shortly, tried to do it again and observed some very unstable/jittery movement of the motor, so something is a miss likely, sometimes I also get the wrong direction error when starting calibration routine. 

Need to do a hardware test or self-test and find out if everything is operating correctly. You dont happen to have something like that ready, for example checking the H bridge channels or whatever else may be required.

SloMusti, 2016-09-26:
Posting raw prints and the calibration codes, they go up about 0.04-0.05 per step. 

Stepper motor: http://www.omc-stepperonline.com/3d-printer-nema-17-stepper-motor-59ncm84ozin-2a-17hs192004s-p-18.html
Magnet: https://www.supermagnete.de/eng/disc-magnets-neodymium/disc-magnet-diameter-6mm-height-2mm-neodymium-n45-nickel-plated_S-06-02-N

Any thoughts?

`DIAAGC (0x3FFC)   1110000000111101100  LF   
ERRFL (0x0001)   1110000000000000000 
11908
11947
11951
11921
11945
11922
11928
11930
11921
11911
11890
11905
11913
11935
11964
11970
12001
12030
12074
12110
12139
12152
12151
12199
12227
12274
12300
12345
12326
12339
12370
12377
12398
12435
12425
12425
12390
12402
12427
12447
12461
12465
12459
12429
12400
12374
12376
12392
12413
12420
12378
12340
12317
12292
12292
12281
12256
12222
12219
12180
12184
[... 321 lines cut ...]
12
20
-31
-74
7
5
-42
-39
-17
29
-17
-83
-17
-18
4
-37
-46
-32
-24
-33
-16
 
newLookup:
 
0.00 , 0.04 , 0.09 , 0.13 , 0.18 , 0.23 , 0.27 , 0.32 , 0.36 , 0.41 , 0.46 , 0.50 , 0.55 , 0.59 , 0.64 , 0.69 , 0.73 , 0.78 , 0.83 , 0.87 , 0.92 , 0.96 , 1.01 , 1.06 , 1.10 , 1.15 , 1.19 , 1.24 , 1.29 , 1.33 , 1.38 , 1.43 , 1.47 , 1.52 , 1.56 , 1.61 , 1.66 , 1.70 , 1.75 , 1.80 , 2.25 , 2.69 , 3.15 , 3.60 , 3.66 , 3.71 , 3.78 , 3.84 , 3.90 , 3.95 , 4.02 , 4.08 , 4.14 , 4.19 , 4.25 , 4.31 , 4.38 , 4.44 , 4.50 , 4.56 , 4.62 , 4.67 , 4.73 , 4.80 , 4.85 , 4.92 , 4.97 , 5.04 , 5.10 , 5.16 , 5.22 , 5.28 , 5.33 , 5.39 , 5.47 , 5.55 , 5.62 , 5.70 , 5.77 , 5.84 , 5.92 , 5.99 , 6.07 , 6.14 , 6.22 , 6.30 , 6.37 , 6.45 , 6.52 , 6.59 , 6.67 , 6.74 , 6.82 , 6.89 , 6.97 , 7.05 , 7.12 , 7.20 , 7.27 , 7.35 , 7.43 , 7.51 , 7.59 , 7.66 , 7.74 , 7.82 , 7.90 , 7.98 , 8.06 , 8.13 , 8.21 , 8.29 , 8.37 , 8.45 , 8.53 , 8.60 , 8.68 , 8.76 , 8.84 , 8.92 , 9.00 , 9.30 , 9.60 , 9.89 , 10.20 , 10.50 , 10.79 , 11.69 , 12.60 , 12.79 , 12.99 , 13.20 , 13.40 , 13.60 , 13.79 , 13.99 , 14.19 , 14.40 , 14.58 , 14.75 , 14.94 , 15.12 , 15.29 , 15.48 , 15.66 , 15.83 , 16.02 , 16.19 , 16.28 , 16.37 , 16.45 , 16.54 , 16.62 , 16.71 , 16.79 , 16.88 , 16.97 , 17.05 , 17.14 , 17.22 , 17.31 , 17.40 , 17.48 , 17.57 , 17.65 , 17.74 , 17.82 , 17.91 , 18.00 , 18.12 , 18.24 , 18.36 , 18.48 , 18.60 , 18.71 , 18.84 , 18.95 , 19.08 , 19.20 , 19.32 , 19.44 , 19.56 , 19.68 , 19.79 , 20.02 , 20.25 , 20.47 , 20.70 , 20.92 , 21.15 , 21.37 , 21.59 , 21.68 , 21.76 , 21.84 , 21.92 , 22.00 , 22.09 , 22.17 , 22.25 , 22.33 , 22.41 , 22.49 , 22.58 , 22.66 , 22.74 , 22.82 , 22.90 , 22.99 , 23.07 , 23.15 , 23.23 , 23.31 , 23.40 , 23.46 , 23.52 , 23.58 , 23.64 , 23.71 , 23.77 , 23.83 , 23.89 , 23.95 , 24.02 , 24.08 , 24.14 , 24.20 , 24.26 , 24.33 , 24.39 , 24.45 , 24.51 , 24.57 , 24.64 , 24.70 , 24.76 , 24.82 , 24.88 , 24.95 , 25.01 , 25.07 , 25.13 , 25.20 , 25.49 , 25.80 , 26.09 , 26.40 , 26.70 , 27.00 , 27.05 , 27.11 , 27.17 , 27.23 , 27.29 , 27.34 , 27.40 , 27.46 , 27.52 , 27.58 , 27.63 , 27.69 , 27.75 , 27.81 , 27.87 , 27.92 , 27.98 , 28.04 , 28.10 , 28.16 , 28.21 , 28.27 , 28.33 , 28.39 , 28.45 , 28.50 , 28.56 , 28.62 , 28.68 , 28.74 , 28.80 , 28.86 , 28.92 , 28.98 , 29.04 , 29.11 , 29.17 , 29.23 , 29.29 , 29.35 , 29.42 , 29.48 , 29.54 , 29.60 , 29.66 , 29.73 , 29.79 , 29.85 , 29.91 , 29.97 , 30.04 , 30.10 , 30.16 , 30.22 , 30.28 , 30.35 , 30.41 , 30.47 , 30.53 , 30.59 , 30.64 , 30.68 , 30.72 , 30.76 , 30.80 , 30.84 , 30.88 , 30.92 , 30.96 , 31.00 , 31.05 , 31.09 , 31.13 , 31.17 , 31.21 , 31.25 , 31.29 , 31.33 , 31.37 , 31.41 , 31.45 , 31.49 , 31.54 , 31.58 , 31.62 , 31.66 , 31.70 , 31.74 , 31.78 , 31.82 , 31.86 , 31.90 , 31.95 , 31.99 , 32.03 , 32.07 , 32.11 , 32.15 , 32.19 , 32.23 , 32.27 , 32.31 , 32.35 , 32.39 , 32.44 , 32.49 , 32.55 , 32.59 , 32.64 , 32.69 , 32.74 , 32.80 , 32.84 , 32.89 , 32.94 , 32.99 , 33.05 , 33.09 , 33.14 , 33.19 , 33.24 , 33.30 , 33.34 , 33.39 , 33.44 , 33.49 , 33.55 , 33.59 , 33.64 , 33.69 , 33.74 , 33.80 , 33.84 , 33.89 , 33.94 , 33.99 , 34.05 , 34.09 , 34.14 , 34.20 , 34.26 , 34.32 , 34.38 , 34.44 , 34.51 , 34.57 , 34.63 , 34.69 , 34.75 , 34.82 , 34.88 , 34.94 , 35.00 , 35.06 , 35.13 , 35.19 , 35.25 , 35.31 , 35.37 , 35.44 , 35.50 , 35.56 , 35.62 , 35.68 , 35.75 , 35.81 , 35.87 , 35.93 , 36.00 , 36.13 , 36.27 , 36.41 , 36.55 , 36.69 , 36.83 , 36.96 , 37.10 , 37.24 , 37.38 , 37.52 , 37.66 , 37.80 , 39.59 , 39.63 , 39.67 , 39.71 , 39.75 , 39.78 , 39.82 , 39.86 , 39.89 , 39.93 , 39.97 , 40.01 , 40.05 , 40.08 , 40.12 , 40.16 , 40.19 , 40.23 , 40.27 , 40.31 , 40.34 , 40.38 , 40.42 , 40.46 , 40.50 , 40.53 , 40.57 , 40.61 , 40.64 , 40.68 , 40.72 , 40.76 , 40.80 , 40.83 , 40.87 , 40.91 , 40.94 , 40.98 , 41.02 , 41.06 , 41.10 , 41.13 , 41.17 , 41.21 , 41.25 , 41.28 , 41.32 , 41.36 , 41.40 , 41.46 , 41.52 , 41.59 , 41.65 , 41.72 , 41.78 , 41.85 , 41.91 , 41.97 , 42.04 , 42.10 , 42.17 , 42.23 , 42.30 , 42.36 , 42.42 , 42.49 , 42.55 , 42.62 , 42.68 , 42.74 , 42.81 , 42.87 , 42.94 , 43.00 , 43.07 , 43.13 , 43.19 , 43.23 , 43.27 , 43.31 , 43.35 , 43.39 , 43.42 , 43.46 , 43.50 , 43.54 , 43.58 , 43.62 , 43.65 , 43.69 , 43.73 , 43.77 , 43.81 , 43.85 , 43.88 , 43.92 , 43.96 , 44.00 , 44.04 , 44.08 , 44.11 , 44.15 , 44.19 , 44.23 , 44.27 , 44.31 , 44.34 , 44.38 , 44.42 , 44.46 , 44.50 , 44.54 , 44.57 , 44.61 , 44.65 , 44.69 , 44.73 , 44.77 , 44.80 , 44.84 , 44.88 , 44.92 , 44.96 , 45.00 , 45.06 , 45.13 , 45.20 , 45.27 , 45.34 , 45.41 , 45.48 , 45.55 , 45.62 , 45.69 , 45.76 , 45.83 , 45.90 , 45.96 , 46.03 , 46.10 , 46.17 , 46.24 , 46.31 , 46.38 , 46.45 , 46.52 , 46.59 , 46.66 , 46.73 , 46.80 , 46.84 , 46.88 , 46.92 , 46.96 , 47.00 , 47.04 , 47.08 ,  [... cut ...]  , 355.95 , 356.02 , 356.09 , 356.17 , 356.24 , 356.32 , 356.40 , 356.45 , 356.50 , 356.56 , 356.61 , 356.67 , 356.72 , 356.78 , 356.83 , 356.89 , 356.94 , 357.00 , 357.05 , 357.10 , 357.16 , 357.21 , 357.27 , 357.32 , 357.38 , 357.43 , 357.49 , 357.54 , 357.60 , 357.65 , 357.70 , 357.76 , 357.81 , 357.87 , 357.92 , 357.98 , 358.03 , 358.09 , 358.14 , 358.20 , 358.31 , 358.42 , 358.53 , 358.65 , 358.76 , 358.87 , 358.98 , 359.09 , 359.21 , 359.32 , 359.43 , 359.55 , 359.66 , 359.77 , 359.88 ,  `

jcchurch13, 2016-09-26:
I think I found your issue: you are using an axially magnetized magnet.  You need a diametrically magnetized magnet.   This one works well, but there are other options:
http://www.kjmagnetics.com/proddetail.asp?prod=D42DIA

The encoder is essentially a two-axis magnetometer that detects magnetic field orientation.  An axially magnetized magnet's field is symmetric about the rotating axis, so the encoder will not be able to detect a rotation!  The raw counts that are printed during the cal routine are the encoder readings at each full step.  They should increment by roughly 82 per step and wrap around at 16384 (2^14).  Note that yours do not change much.

SloMusti, 2016-09-27:
Noted, I have tried now with https://www.supermagnete.de/eng/sphere-magnets-neodymium/sphere-magnet-diameter-5mm-neodymium-n35-nickel-plated_K-05-N spherical magnet, although a bit trickier to install, gives a full range of values on the encoder. 

One step closer, but not creating stable operation. Upon closer observation it appears some steps are already lost in the calibration procedure, so it is time to double-check the hardware and/or wait for kickstarter boards to arrive to compare and test.

bretskee, 2016-10-20:
Hi,

I have exactly the same issue. I too have built a Mechaduino. It seems to be fine in that it steps and calibrates. On power up, the motor is fixed in position with lots of resistance. As soon as I enable closed loop mode, I can spin the motor freely, the led lights and nothing happens. It has at times become unresponsive to further serial commands. A reset brings it back to life. My motor is a 0.9 degree motor. I have changed the spr value with no effect. I have not looked into it deeply, but there are some other numbers hard coded as '200' that look suspiciously like they should be defined as spr. I have not looked closely as my calibration numbers but I will as soon as I get a little time. I have verified however that I get reasonable numbers from my encoder and I have definitely got a correctly oriented magnet.

OKRobotics, 2016-11-19:
I have received 3 of the boards that I ordered from Tropical Labs. I am using the magnets that came with them. I am trying to use them with some 0.9 degree stepper motors. They work with the 1.8 degree motors, and I have changed spr to 400 in parameters.cpp. Is there something else that I need to change in the code to get these boards to work with the 0.9 degree motors? They will calibrate, and move about 1 degree, then fault out.

closed 2017-01-05 by jcchurch13