jcchurch13/Mechaduino-Firmware #6: Configuration defaults for different steppers/magnets

SloMusti, 2016-09-26:
Collect and document defaults for common configurations as a guide for assembling custom motor configurations.

I propose the following format or similar:

- motor type:
- magnet type:
- magnet to encoder distance
- PID values
- floatangle (not quite clear why/how different)
- any other variables possibly required

@jcchurch13 Can you please post your current working values for stepper motors and magnets you have.

minid33, 2016-10-29:
Additionally, I'd appreciate some help with steps/mm.

I have a smoothieboard, I was able to fix the stepper direction by reversing the pin in the config but when it comes to adjusting the steps/mm in the config it seems to make no difference, which leads me to believe this is a setting for the built in stepper drivers im bypassing.

I tried changing parameters.ccp dir to 0 and this did not reverse the direction of the stepper. The names of variables in this project could benefit from being a little more verbose. 

How can I adjust the equivalent of steps/mm in this code? I was messing with spr in parameters.cpp and this seemingly had no effect.

jcchurch13, 2016-10-29:
I presume you have uncommented this line in the main loop to use the step/dir interface:
"r=0.1125*step_count" :
r is the control loop setpoint variable.  If you are in closed loop position mode, setting r = 45.3  will cause the Mechaduino to point to 45.3.  
step_count is the running count of step commands received via the step/dir interface.  0.1125 is just the scale factor (0.1125 degrees per step)... this value happens to be 1.8/16, but you can change it to whatever you like.
-Joe

minid33, 2016-10-30:
That did the trick, thanks.

closed 2017-01-05 by jcchurch13