jcchurch13/Mechaduino-Firmware #9: Firmware Branches

Eriobis, 2016-10-22:
Hi,
Since the mechaduino have been released to public, it would be nice that your git tree contains the working firmware in the working branch, or maybe a tag with 0.1 written on it.
I had a hard time making it work ( i tried 3 steppers ) to make it work, until I used the developement branch.

jcchurch13, 2016-10-22:
Hi, 
We have been keeping the "working" firmware in the multi-file branch which is set as the default.  The development branch is being used for development, so it is a little less stable.  If the multi-file branch is not working for you, let us know.
-Joe

Eriobis, 2016-10-23:
The motor was struggling and when I tried some commands everything disconnected. So I tried the developement branch.
Everything is smooth. Except 2 bugs, mayve I can open issues for that. Sometimes ( 1 on 3 ) times, the Y ( enable interrupt ) makes everything stop and the code just stop working.

Second, the "0" of the motor cannot be reached easily, it is oscilliating constantly even with a lower P or a less agressive Derivative.
I will let you know my developement as I make tests !

oddball1, 2016-10-29:
I dont understand how to download the Mechaduino 0.1 firmware to Arduino?. I purchased 3 of these stepper motors with the encoders and I'm stuck at this point, if i copy and paste the one file that says Mechaduino 0.1 and compile it, it says i'm missing the utils.h and errors out. Please any input would be appreciated as i am new to this.

minid33, 2016-10-29:
Try updating your version of the arduino ide @oddball1 I had an older version and it said I was missing stuff.

oddball1, 2016-10-29:
Thanks for the fast responce minid33, I am using the latest version 1.6.12, I think I'm not doing it correctly.
I can see the file, utils.h on the same page as the Mechaduino 0.1, but not sure if i'm supposed to include it some how. I have also tried anouther version of Arduino 1.6.10, it was the same result.

jcchurch13, 2016-10-29:
Hi oddball1,
Please download the entire repository from here (click "clone or download"):
[https://github.com/jcchurch13/Mechaduino-Firmware](url)

Then you will need to unzip the file and navigate to the Mechaduino_01 file.  Double clicking the Mechaduino_01 file should open all of the files in the Arduino IDE (There should be 11 tabs/files).

oddball1, 2016-10-31:
Thanks jcchurch13, thats exactly what I was doing wrong, I've never used this web page before so I had no idea how to down load this, Thanks again, very much appreciate your help.

oddball1, 2016-10-31:
Oh darn, spoke to soon, well not really, I don't get the same error ad can see all the tabs in Arduino,but now I get these errors, Arduino: 1.6.12 (Windows 10), Board: "Arduino/Genuino Zero (Programming Port)"

Sketch uses 98,608 bytes (37%) of program storage space. Maximum is 262,144 bytes.
Open On-Chip Debugger 0.9.0-gd4b7679 (2015-06-10-22:24)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

Invalid library found in C:\Users\Darrell\Documents\Arduino\libraries\Marlin-Marlin_v1: C:\Users\Darrell\Documents\Arduino\libraries\Marlin-Marlin_v1
Invalid library found in C:\Users\Darrell\Documents\Arduino\libraries\Marlin-Marlin_v1: C:\Users\Darrell\Documents\Arduino\libraries\Marlin-Marlin_v1

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bretskee, 2016-11-01:
Just quickly looked at you post and in board selection you should be using 'native USB port' instead of 'programming port'.

oddball1, 2016-11-03:
Thanks again for your help, just back to it tonight myself, so that did work, I was able to upload the firmware. I restarted my computer, checked the port it was in (com7) opened the serial on the called for baud rate and typed s and send, nothing, tried others like d and w, all nothing. I have a 5vdc power supply at 2amps, also measured voltage, should there be a power light on the board? 
I was really hoping this would be a lot easier to do, I have this feeling I'm going to spend more time trying to get this up and running, I hope I'm wrong, when I saw the video on Kickstarter on how one stepper motor could control another one (like Simon says) this is why I backed the project, that's all I'm try to set up, I turn one stepper and the other follows exactly those steps. I am working on a product for patenting and could see your product being part of my product, in the thousands if not tens of thousands in the future, but really need to get this part up and running, any support you could offer would be greatly appreciated, thanks again for your time. Darrell Stein.

oddball1, 2016-11-06:
Does anyone have any sugestions on where i go form here? I can't seem to get these motors running, just paper weights at this time.

jcchurch13, 2016-11-06:
Hi oddball,

Can you give us any more information?  It sounds like you were able to successfully upload the firmware but you haven't been able to communicate over serial?  I'd try resetting/disconnecting and reconnecting the board a couple times.  You should be able to get a response from the board even without an external power supply.  You will need 8V to 25V on the screw terminals to provide power to the motor to get it to move though.  You only need to provide 5V to Vin if you are not using the USB.  (See the wiring diagrams here: http://tropical-labs.com/index.php/mechaduino)
The LED is not a power LED, you can program it to do whatever you like.
-Joe

oddball1, 2016-11-07:
Thanks jcchurch13 again, I'm just not sure what to do with these motors and boards, it just doesnt seem to be very easy to get them to work, I've had them for a bit now and can only get one step ahead every few days, I feel clearer instructions would certainly be helpful for those that are new to this type of system.
So now i have 12vdc to the board and the motor locks up now when its pluged into the usb port, but when i bring up the serial monitor and type "s" and enter, nothing happens, motor is still locked, i'm assuming i should have the mechaduino 01 program running in the arduino IDE. I have unpluged it several times, restarted the program and even tried all 3 motors and boards, and yes it seems that the firmware was successfully uploaded.

Electroradiography, 2016-11-07:
I have exactly the same issue with three motors. Is there a solution yet?

Bob

oddball1, 2016-11-08:
I guess not yet

jcchurch13, 2016-11-08:
Are you using the latest version of the "multi-file" branch?  Is it possible that "enableTCInterrupts()" is uncommented in the main file?  That's the only thing I can think of that might cause this behavior.  Sorry guys,  I haven't been able to replicate this issue, so I'm not really sure what else to suggest.

oddball1, 2016-11-08:
I hope so, it's the one you just sent me about a week ago.

Electroradiography, 2016-11-08:
Here are things I noticed when learning how to set up the Mechaduino:

It's not clear to me whether my boards have a bootloader. I ordered mine
from the Tropical

Labs website, not Kickstarter.

Instructions say to make sure to have the latest bootloader but don't tell
how to check.

After installing "Arduino SAMD (32-bits ARM Cortex-M0+)" you are to connect
the usb cable and

Mechaduino will appear as Arduino Zero. Is that under Ports in the Device
Manager?

When the link is followed to Github to download the latest Mechaduino_01
multi-file firmware

you arrive at a page that has no download option. If you back up two
branches to Mechaduino-

Firmware there is a download option. That's what I used. It captures more
than Mechaduino_01.

Is that correct?

The com port seems to change almost at random when you are working the
instructions. It's

important to keep an eye on the assignment to make sure you have the right
one assigned.

Is it necessary to change the driver speed to match the speed assigned to
the monitor?

When s is typed in the serial monitor more than twice my serial monitor
hangs. The usb cable

must be disconnected to clear it.

Which port should be selected, the programming port or the native usb port?

When the Mechaduino reset is pressed twice, the device manager displays the
port as

bootloader. What's the significance?

I'm using 24v to power the board and have verified it with a dvm. Correct
polarity has been

checked.

Once the Mechaduino_01 code has been loaded the motor shaft locks up and a
slight noise comes

from the board.

Here's the commented line: // enableTCInterrupts();     //start in closed
loop mode

Are the Github instructions all that's needed to get going or did I miss
something?

jcchurch13, 2016-11-08:
Hi all,

I think I may have found the issue:  "SerialCheck" needs to be uncommented in the main loop.  I've updated the github repo.  Thank you for helping me find this issue.  Please try the updated firmware.
-Joe

jcchurch13, 2016-11-08:
Electroradiography,

Thank you for all your feedback.  I'm doing my best to provide clear documentation for the hardware/firmware.  Right now I'm working very hard to make sure everyone receives their hardware, so I have been a little slow to make updates.  Let me try to answer some of your questions.

For the latest, most up-to-date instructions on how to configure your Mechaduino, please check our [hackaday.io](https://hackaday.io/project/11224-mechaduino) page.  In particular, check out the [instructions](https://hackaday.io/project/11224/instructions) section.  

The latest firmware can be found here in our github repository.  Yes, you do need to navigate up and download the whole repository for now.

We’ve also created a [google group](https://groups.google.com/forum/#%21forum/mechaduino) for users to share technical info. 

All Mechaduinos ordered from Tropical Labs (Kickstarter & our website) should have bootloaders.  If your Mechaduino appears as a COM port, it has a bootloader.

Please select "Arduino/Genuino Zero (Native USB Port)"  under boards manager. 

Yes, the COM port may change if the Mechaduino is reset or disconnected.  This seems to be an issue with all SAMD based arduinos (Zero and Due type boards).  

You should adjust the baud rate in your serial monitor to match the Mechaduino (115200 by default).

Pressing reset twice causes an Arduino Zero to enter bootloader mode.  All this really does is stop any firmware that you uploaded to the board from running (helpful if you want to upload new firmware without the board trying to do anything).

The motor "locking up"  is the phase current being applied to the motor.  By default, the Mechaduino starts up in an open loop stepping mode.

I hope that helps,

-Joe

oddball1, 2016-11-08:
Thanks jcchurch13, could you please send the link!, when I go to the site I can't seem to find the download area, but when you sent the link last time it was clear where it was, thanks again.

oddball1, 2016-11-08:
Thanks, I'll try to make some time tonight to try this. I will be looking for instructions on how to hook up two motors to control each other, like in your kickstarter video, when you turn one the other follows, just a heads up, thanks again, I know you have a lot on the go right now so I do appreciate this.

Electroradiography, 2016-11-08:
That did it - Thanks Joe

oddball1, 2016-11-08:
me too, i was able to run the calibration, but unable to copy the 16 thousand plus numbers, i can get them highlighted as per the instructions, but right clicking on the mouse wont bring up the copy command and the short cut on the keyboard of Ctrl + C will not copy either?? any sugestions ?

oddball1, 2016-11-09:
ok, i guess ctrl c did copy it, and i got it into the parameters.cpp, ive calibrated all 3 motors now, woot woot.
Now my goal is to get one stepper to be able to control anouther stepper just like the video on Kickstarter, ideally i would like to control the movements of 6,8 or more stepper motors with one stepper as the master, so if the master was at say 18 degrees, the slaves would be the same. Any ideas on this anyone!

oddball1, 2016-11-12:
Now my goal is to get one stepper to be able to control anouther stepper just like the video on Kickstarter, ideally i would like to control the movements of 6,8 or more stepper motors with one stepper as the master, so if the master was at say 18 degrees, the slaves would be the same. Any ideas on this anyone!

jcchurch13, 2016-11-13:
Oddball1,
Please see the description in this video:  https://www.youtube.com/watch?v=rSugkwWcFOY
There are a number of ways that you can implement this.  The simplest way that I have found is to feed the current position of each motor into the other motor and use it as the setpoint of the position control loop.  You can use analog I/O pins to do this (but there are other ways too..I2C for example).  This set up will definitely involve some tuning/coding to get good performance.  Sorry, that's all the help I can offer at the moment.  I am very busy working on general firmware improvements.
Hope that helps,
Joe

oddball1, 2016-11-17:
Thanks, I really don't understand how to do this.

Electroradiography, 2016-11-17:
I need to use two 5v input signals to drive the motor. One signal will command clockwise motion for a few seconds then stop. The second signal will do the exact opposite. I can make the motor respond with the serial monitor but I'm not sure how to code my application. I'm afraid I will damage the Mechaduino code unless I know more about how it works. Is there a guide available to help me?

jcchurch13, 2016-11-18:
We're working on putting together some more documentation.  Please stay tuned!
-Joe

oddball1, 2016-11-19:
That would be nice, then I could use them,  hopefully you will include how to have one drive another like in your video, I egearly await .

closed 2017-01-05 by jcchurch13

oddball1, 2017-04-24:
Any documintations yet on hooking up two motors like in the video I saw when I backed this project on kick starters?? You show one motor controlling another one, back in January you mentioned you would be putting something together

jcchurch13, 2017-04-28:
To accomplish the electronic gearing shown in the video, you can have the position loop setpoint controlled by an analog input on both motors, and set an analog output pin proportional to the encoder reading on each motor.  Then just feed the analog position output from each motor into the setpoint input of the other.  Please see the Mechaduino Manual that we added to in the repo for examples of how to configure inputs/outputs/control loop setpoints.  This configuration will definitely require some tuning/filtering depending on how you want it to behave.  I haven't published a detailed description of how to do this yet because though conceptually simple, it takes a certain amount of experimentation depending on exactly how you want it to behave.  Also, I think there may be better, more sophisticated ways to set this up (using, say I2C instead of analog setpoints), but I haven't had time to develop this.  We've been focusing on trying to keep the hardware available and make general improvements to the firmware.  I encourage you to experiment with the set up I described above.  It's not that hard to get it up and running, but its not as simple as just running a certain piece of code.