jcchurch13/Mechaduino-Firmware #10: USB serial communications lock up

rreilink, 2016-10-25:
I just received my Kickstarter Mechaduinos. I have flashed the Mechaduino_01 firmware. Using the serial monitor, I can give a few commands to make it move (I did perform the calibration first). However, after a few commands the communication locks up. Actually, it causes even the Arduino PC application to hang. The only thing to make it work again is unplug/replug the USB connector.

The issue occurs both in Mac OS X and in Windows 7. On Mac OS X I have used both the Arduino PC application and Minicom as serial terminal applications. In all cases I get the same result. Is this a known firmware issue?

ericmoderbacher, 2016-10-27:
BUMP.  I experienced the same issue on Mac OS X.  Have not tried windows.

Eriobis, 2016-10-29:
Same issue here, I've found that if I enable the interrupt before choosing the mode, everything stalls ( the communication )

ericmoderbacher, 2016-11-08:
is this fixed by the "SerialCheck must be uncommented in loop" check in?[https://github.com/jcchurch13/Mechaduino-Firmware/commit/59ae0f516e6fb6f75cfc882bbd4a4016c0681eee](url)

jcchurch13, 2016-11-09:
I'm not sure.  It sounds like you are able to get the Serial monitor to work up to a certain point.  (That fix was for a completely unresponsive serial monitor.)  Can you provide some more details about when it stalls?  The commands 's' and 'd' will not work properly if you are in closed loop mode (interrupt enabled).

ericmoderbacher, 2016-11-12:
I had some time to try out the mechadinos again.  I'm not experiencing the communications issue anymore.  Thanks!

closed 2017-01-05 by jcchurch13