jcchurch13/Mechaduino-Firmware #13: Feature request: PID Autotuning

minid33, 2016-10-30:
I'm just about to kick off a print for the first time and I can't help but feel like my PIDs are going to be innacurate, maybe even hugely inacurate. 

PID autotuning is an obvious answer to this problem to get into the ballpark of a good tune. I'd love this to be implemented in future releases.

minid33, 2016-11-26:
Discussed further here, https://groups.google.com/forum/#!topic/mechaduino/29lYo7xKL8Q
Has been implemented in a fork

Kaiwol1990, 2017-01-10:
Hi minid33,

it's not a real fork anymore. I've changed quiet a bit in my firmware version, maybe give it a try and send me some feedback.
You can also use the PID setting you get with my firmware but have to recalculate them for this firmwareversion. To do that you have to considere the different frequencys. Here's how you can calculate them from my firmware to the this one.

Kp = Kp_my_version
Ki = Ki_my_version * (FPID / FPID_my_version)
Kd = Kd_my_version* (FPID_my_version / FPID)

if you take a look at my firmware you can also use the dev branch. I've implemented filtering, feedforward, friction compensation and inertia copensation. But there's no autotune for those variables.
If your familiar with such things give it a try.

- Kai