jcchurch13/Mechaduino-Firmware #14: Configuration to work with 0.9 degree steppers.

OKRobotics, 2016-11-19:
I have been reading through all the other posts, and it looks like these boards should be able to drive 0.9 degree steppers. The only information that I have found about it is that spr needs to be set to 400 instead of 200 in the Parameters.cpp file. I have done that, and the motor now runs through all 400 steps during calibration, and will move fine in open loop mode, but when put into closed loop mode, it fails. The same board and magnet works fine on a 1.8 degree stepper with spr set to 200. Any ideas?

Apehand, 2016-11-22:
Hey there.
I'm running in the same problem. Do you have any solutions jet? Greetings.

jcchurch13, 2016-11-22:
Hi,
There are a couple of numbers that will need to be changed in the output() function in the utils.cpp file.  I'm working on an updated version of the firmware that should fix this issue.
-Joe

Apehand, 2016-11-23:
Hey Joe, do you have any values for testing? It is a quite annoying problem.

jcchurch13, 2016-11-23:
I've updated the firmware and confirmed that it works on a 0.9 degree stepper.  All you need to do is set spr = 400 in parameters.cpp, and set PA = 0.9 in State.cpp.

-Joe

Apehand, 2016-11-25:
I tested it today and I think it is now working properly. Thank you a lot. It is a amazing work!

closed 2016-11-30 by jcchurch13