jcchurch13/Mechaduino-Firmware #17: Missing file include and invalid type.

Razaekel, 2016-11-27:
Controller.cpp has a '#include <SPI.h>' but that include is not included in Mechaduino_01.ino, which gives an error on compiling.

After adding '#include "SPI.h"' to the .ino, it gives a ''PROGMEM' does not name a type' error in Paramters.h.

Razaekel, 2016-11-27:
After removing the references to PROGMEM, it  gave another error about a missing "Wire.h". After adding this include to the .ino, it compiled fine.

mikeanton, 2016-11-27:
I'm pretty sure I had to upgrade to the most recent Arduino IDE (1.6.12), and then it compiled fine.  I had problems like this as well using a slightly older version (1.6.5).

Razaekel, 2016-11-27:
@mikeanton You are correct, updating to the latest Arduino IDE (1.6.13) allows it to compile. 

This IDE version requirement should be mentioned in the README.

Closing this issue, as it's no longer valid.

closed 2016-11-27 by Razaekel