jcchurch13/Mechaduino-Firmware #21: Weird issue with driver heating up and onboard led turning off.

RobWks, 2017-02-28:
Hey everyone,

So, finally got around to setting up two mechaduino's and I've run into the same issue with both. I can run the calibration for the mechaduino fine and paste the lookup values to Parameters.cpp -> const float lookup[] ={}. For testing I commented out `digitalWrite(ledPin,LOW);         // turn LED off ` so that the LED would remain on. I can move the stepper around with single steps using 's', but when I go to closed loop position mode by sending 'x' then 'y' over the serial monitor, the led turns off when I send 'y' and the motor driver gets too hot to touch for longer than a couple seconds. When I try to send 'r' to enter a new step point, it asks me to "Enter setpoint:", which I can do and it prints the value entered back at me, but the stepper doesn't move for any input. Is the LED going out an indication of some sort of issue? Is something shorting?  Any help is appreciated!

jcchurch13, 2017-02-28:
This sounds like an issue with the cal tables.  Can you double check that your cal tables consist of steadily increasing values.  Each entry should increase by roughly 0.02, and the table will wrap around to zero once it reaches 360.   If yours does not do this, try re-calibrating.  If the problem persists let me know.

RobWks, 2017-02-28:
Hey Joe, yes looks like that was the issue. Looking through the cal table I noticed that each entry was increasing by roughly 0.04. Made me realize that the random stepper I grabbed was actually 400 steps/rev instead of 200 steps/rev. Fixed the spr variable to be 400 and that fixed it. Thanks a lot!

closed 2017-02-28 by RobWks