jcchurch13/Mechaduino-Firmware #38: Latest code does not compile

AThilenius, 2019-10-18:
Starting with [this commit](https://github.com/jcchurch13/Mechaduino-Firmware/commit/055bc004133c3f7c70fbc5b1394e09163140d0f2) the `FlaggedStorage` library was used, resulting in the compilation error: `FlashStorage.h: No such file or directory` on a clean Arduino install.

Repro steps:
- From a clean Arduino install
- Add `Arduino SAMD Boards (32-bit ARM Cortex-M0+)` in the Board Manager
- Git clone `master`
- Open `Mechaduino\Mechaduino\Mechaduino.ino`
- Select any `Arduino Zero` (doesn't matter, all fail)
- Hit `Verify`

Given that the README says "New firmware is in the default "master" branch" I think it's expected that the master branch be green (would recommend some CI).

Latest `Release` still works.

charles-blouin, 2020-10-05:
Temporary workaround: Sketch>Include Library> Manage Library

Look for the library FlashStorage, and install version 0.7.1

mmwolbrink, 2020-10-06:
> 
> 
> Temporary workaround: Sketch>Include Library> Manage Library
> 
> Look for the library FlashStorage, and install version 0.7.1

Thats not a workaround, you are supposed to do that anyway.