jcchurch13/Mechaduino-Firmware #43: Anti-cogging, proof on concept

charles-blouin, 2020-10-18:
Hello there,

I made a test implementation of anticogging. Here is a [video](https://youtu.be/VyuWUg4WjfY). It is especially noticeable for fine control in velocity and torque mode.

The calibration is done in python for now. It should be migrated to the firmware, similarly to the encoder calibration.

The calibration is longer, as it requires finer steps. It should be done if forward and reverse direction, and averaged, as there is a constant friction term. The blue line is a calibration pass in both directions.

[image]

[image]

Changes to mechaduino:
https://github.com/charles-blouin/Mechaduino-Firmware/commits/cogging
Calibration code
https://gitlab.com/charles-blouin/pendulum-ml/-/tree/master/anticogging_calibration

 You can run 'measure-torque-ripple' followed by interpolate_360

ChristianFrisson, 2020-10-26:
Hi @charles-blouin, anti-cogging is a very interesting feature for Mechaduinos and your [video](https://youtu.be/VyuWUg4WjfY) is compelling!

I can access your [github fork](https://github.com/charles-blouin/Mechaduino-Firmware/commits/cogging), but I can not reach your [gitlab repository](https://gitlab.com/charles-blouin/pendulum-ml/) (error 404 Not Found). Could you please check?

Kind regards,
Christian

charles-blouin, 2020-10-26:
@ChristianFrisson Sorry about this, it is fixed.

It may be possible to reuse the rotary encoder calibration code for an anticogging calibration routine. The hardware may have just enough RAM. Right now, the encoder is calibratedwith 200 points in an array named fullStepReadings[spr]. For the anticogging, we need a little over 360*4 points. The calibration could be done in two passes, so the full calibrated array with 16k points would be updated after the second calibration pass.

mmwolbrink, 2021-02-03:
This Sparked my interest, I'm going to see if I can integrate this into mechaduino firmware. I will keep you guys posted.